toyota-connected/ivi-homescreen #56: Mouse pointer motion events from wayland to flutter sometimes violate debug assertion state.is_down

doesnotexist, 2022-06-22:
On my ubuntu 20.04, wayland+weston+ivi-homescreen running the default flutter create "increment" button app with debug engine v2.10.3 arm64,  I'm finding that the app often crashes when using mouse input:

```
[FATAL:flutter/lib/ui/window/pointer_data_packet_converter.cc(189)] Check failed: !state.is_down. 
```

https://github.com/flutter/engine/blob/main/lib/ui/window/pointer_data_packet_converter.cc#L190

I believe this is a manifestation of a bug in translating wayland input events to flutter, specifically within Display::pointer_handle_motion() wayland input event listener
https://github.com/toyota-connected/ivi-homescreen/blob/main/shell/display.cc#L399

The call to getPointerPhase can find that p->buttons is not null and  state == WL_POINTER_BUTTON_STATE_RELEASED which ends up translating the event to  FlutterPointerPhase::kUp.

Based on what I've gathered from reading the comments in embedder.h
https://github.com/flutter/engine/blob/main/shell/platform/embedder/embedder.h#L727-L743

I propose that the wayland input event listener Display::pointer_handle_motion()  should only translate to  FlutterPointerPhase::kMove or  FlutterPointerPhase::kHover  depending on whether the left mouse button is currently pressed down or not during the motion event.   PR that implements this change coming shortly.

Currently I'm reading wayland documentation to improve understanding of pointer events on that side before submitting my PR. https://wayland-book.com/seat/pointer.html

doesnotexist, 2022-06-22:
This bug doesn't happen in touch handlers because a boolean based on counter down_count prevents multiple kUp events from being ever being sent to flutter. 
https://github.com/toyota-connected/ivi-homescreen/blob/main/shell/display.cc#L636

doesnotexist, 2022-06-22:
The linux desktop shell implements what I proposed above: https://github.com/flutter/engine/blob/main/shell/platform/linux/fl_view.cc#L383

jwinarske, 2022-06-22:
PR merged

closed 2022-06-22 by jwinarske