toyota-connected/ivi-homescreen #70: Multi-View Video Player

jwinarske, 2022-07-14:
Only a single instance of Video Player is currently supported.  The static plugin needs to be refactored to handle multiple views.

Currently a low priority.

amrbadr12, 2023-07-17:
Hey! Thanks for reporting that, we currently have an issue somewhat feels related to that but not entirely sure.

We have multiple `VideoPlayerController`'s that are initialised at the same time and when the first video ends and the second one starts playing we get this exception:

 ```
══╡ EXCEPTION CAUGHT BY SERVICES LIBRARY ╞═══════════════════════
The following FormatException was thrown during a platform
message callback:
Invalid envelope

When the exception was thrown, this was the stack:
#0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:609:7)
#1      EventChannel.receiveBroadcastStream.<anonymous closure>.<anonymous closure> (package:flutter/src/services/platform_channel.dart:498:34)
#2      EventChannel.receiveBroadcastStream.<anonymous closure>.<anonymous closure> (package:flutter/src/services/platform_channel.dart:493:47)
#3      _DefaultBinaryMessenger.setMessageHandler.<anonymous closure> (package:flutter/src/services/binding.dart:380:35)
#4      _DefaultBinaryMessenger.setMessageHandler.<anonymous closure> (package:flutter/src/services/binding.dart:377:46)
#5      _invoke2.<anonymous closure> (dart:ui/hooks.dart:190:15)
#9      _invoke2 (dart:ui/hooks.dart:189:10)
#10     _ChannelCallbackRecord.invoke (dart:ui/channel_buffers.dart:42:5)
#11     _Channel.push (dart:ui/channel_buffers.dart:132:31)
#12     ChannelBuffers.push (dart:ui/channel_buffers.dart:329:17)
#13     PlatformDispatcher._dispatchPlatformMessage (dart:ui/platform_dispatcher.dart:589:22)
#14     _dispatchPlatformMessage (dart:ui/hooks.dart:89:31)
(elided 3 frames from dart:async)
```

jwinarske, 2023-07-18:
that looks entirely like a Dart back trace.  What branch and commit are you running on, and what package commit of video player?

Also note multiple video players are not currently supported on main or agl branches.  It requires some re-work of the gstreamer plugin code.

amrbadr12, 2023-07-19:
Good to know thanks,

For ivi-homescreen it's:
https://github.com/toyota-connected/ivi-homescreen/commit/b18a478df377f9ed34e6aaaf364a504961c4e582

For the video player it's:
```
video_player: 2.6.0
video_player_platform_interface: 5.1.4
```

jwinarske, 2024-01-12:
A rework in upcoming release will enable multiplayer scenario