toyota-connected/ivi-homescreen #75: Running on Desktop Ubuntu

bernardcraddock, 2022-07-29:
I can build desktop ubuntu 20 just fine but when it comes to running 
Copy libflutter_engine.so to /usr/local/lib or use LD_LIBRARY_PATH to point downloaded engine for build: 

Question where do i get libflutter_engine.so from ?
Do I try and build a copy using https://github.com/meta-flutter/flutter_embedded  
or 
Can i use one of my local copies created during flutter snap install maybe the release copy 

parallels > find  /home/parallels/snap/flutter -type f -name "libflutter*.so" -print
/home/parallels/snap/flutter/common/flutter/bin/cache/artifacts/engine/android-x86/libflutter.so
/home/parallels/snap/flutter/common/flutter/bin/cache/artifacts/engine/android-x64/libflutter.so
/home/parallels/snap/flutter/common/flutter/bin/cache/artifacts/engine/linux-x64/libflutter_linux_gtk.so
/home/parallels/snap/flutter/common/flutter/bin/cache/artifacts/engine/linux-x64-release/libflutter_linux_gtk.so
/home/parallels/snap/flutter/common/flutter/bin/cache/artifacts/engine/linux-x64-profile/libflutter_linux_gtk.so

please advise
regards

jwinarske, 2022-07-29:
@bernardcraddock For desktop workflow one of my devs is working on a python script that setups a flutter workspace.  It pulls all the parts for running on desktop.  Hoping to published it to meta-flutter by next Wednesday.

jwinarske, 2022-07-29:
The engine for desktop (x86_64) is coming from Flutter archive.

bernardcraddock, 2022-07-29:
@jwinarske - perfect thank you - i'll wait for the script

bernardcraddock, 2022-08-21:
@jwinarske is the  python script done if so where is it?
If not done are you  able to provide me access to manually run your workflow?
 
I'm requesting for 2 reasons

1. Review real time exactly what it's doing and how  - which will be enough for me
2.  Will build ped workflow once i'm comfortable with process

jwinarske, 2022-08-21:
@bernardcraddock As of right now it's 93% feature complete.  You're welcome to start using it.

https://github.com/meta-flutter/meta-flutter/tree/jw/custom-devices/tools

When this branch is gone, it means it's been merged into the kirkstone branch.

If you add your GitHub dev token to "github_token" in the flutter_workspace_config.json file, then the script will download/install flutter-auto, as well as install my meta-flutter AGL QEMU image (integration reference).

The process is run the python script from location where you want the Flutter Workspace.

Sample Usage
```
cd $HOME
mkdir test && cd test
git clone -b jw/custom-devices https://github.com/meta-flutter/meta-flutter
mkdir workspace && cd workspace
 ../meta-flutter/tools/setup_flutter_workspace.py --workspace-cfg ../../meta-flutter/tools/flutter_workspace_config.json 
```
Once this is complete from your development terminal run:
```
source $HOME/test/workspace/setup_env.sh
```
then navigate to your app folder, and run
```
flutter run
```

The last items I'm working out:
1. custom-device working on anything other than master.  Flutter stable needs fix from master to enable.
2. Automate target debug image query of Flutter SDK version.

So running on Ubuntu (x86_64) debugging with flutter-auto is currently functioning.  The caveat is that you need a functioning Vulkan stack on your desktop, which is intentional to prevent loading your host CPU.

If you have any input or feedback on the script please create an issue on meta-flutter so others can benefit from it.

jwinarske, 2022-08-21:
The intent is that an organization tailors the json config to their dev environment.

jwinarske, 2022-08-21:
So once you have your custom `flutter_workspace_config.json`, you create the root workspace folder, and copy your `flutter_workspace_config.json` in.  Then run:
```
curl --proto '=https' --tlsv1.2 -sSf https://raw.githubusercontent.com/meta-flutter/meta-flutter/jw/custom-devices/tools/setup_flutter_workspace.py | python3
```
Once `jw/custom-devices` is merged the link will be:
```
curl --proto '=https' --tlsv1.2 -sSf https://raw.githubusercontent.com/meta-flutter/meta-flutter/kirkstone/tools/setup_flutter_workspace.py | python3
```