toyota-connected/ivi-homescreen #108: embedder API vulkan bug workaround

ardera, 2023-04-07:
Hey,

I just stumbled upon an embedder API bug with vulkan, that's fixed already but still present in flutter stable:

Issue: https://github.com/flutter/flutter/issues/118956
PR: https://github.com/flutter/engine/pull/39051
Problematic code: https://github.com/flutter/engine/blob/3.7.10/shell/platform/embedder/embedder.cc#L560-L561

I had to add a workaround to flutter-pi to make vulkan work again: https://github.com/ardera/flutter-pi/commit/90faaf0866e9c6cd3b97e5743e9af415093ab543

Just thought I'd bring this up to you, not sure you're already aware of it. I didn't see a workaround in your code at least. 🙂

jwinarske, 2023-04-10:
@ardera What engine version?

ardera, 2023-04-10:
The flutter engine repo now has the flutter SDK version tags too, so this is with flutter engine `3.7.10` (which is commit `ec97508`).

jwinarske, 2023-06-19:
Is this still relevant in recent engine builds?  Internally we're on 3.3.7 so it's not a big priority at the moment.

ardera, 2023-06-19:
It's fixed in 3.10: https://github.com/flutter/engine/blob/3.10.5/shell/platform/embedder/embedder.cc#L565-L566

jwinarske, 2023-07-03:
Closing since issue not found in 3.10.5

closed 2023-07-03 by jwinarske