toyota-connected/ivi-homescreen #136: Command Line Option for Bundle Path does not autocomplete in bash

Markus43, 2024-01-11:
I am playing around with multiple apps and embedders via meta-flutter at the moment.

Specifying the path of the bundle works like this with homescreen, the sony embedder and flutter-pi:

homescreen --b=/usr/share/flutter/flutter-gallery/3.16.5/release/
flutter-client -b /usr/share/flutter/flutter-gallery/3.16.5/release/
flutter-pi --release /usr/share/flutter/flutter-gallery/3.16.5/release/

When playing around with multiple test apps or new flutter versions, I often modify this path parameter by hand. The format of flutter-client and flutter-pi allows auto-completion of this path parameter, while the leading '=' of homescreen breaks autocompletion of the path in bash.

Maybe homescreen could add an alternative way of specifying the bundle path that also has a space before the path?

jwinarske, 2024-01-12:
@Markus43 Hi there.  It would be a very low priority at this point.  I would recommend adopting the use of custom-devices.  My workspace automation project manages the install of it for a given device, and is easily customizable to a given development environment.

Markus43, 2024-01-15:
> @Markus43 Hi there. Is this for a desktop use case that requires regular end user interaction?

No, my usecase is completely on embedded. I just noticed this when getting started with custom embedders: As a first step, I just added a few of the example apps in meta-flutter to my yocto build and tried to run them. I just found the missing auto-complete annoying, and so did a colleague of mine who also tried the apps. And this is probably the first thing that many people do when getting started with meta-flutter. So I thought it would make sense to report an issue and make you aware of it. I agree that it's low priority, as for the real development work I'll of course switch to custom devices where all of that is automated.

Markus43, 2024-06-28:
Tested this again today, after updating my yocto build to scarthgap. And completion works for me now, both via wayland terminal and when connected remotely via ssh. It looks like something has changed in these shells in-between, making this issue obsolete.

closed 2024-06-28 by Markus43

jwinarske, 2024-07-09:
https://github.com/toyota-connected/ivi-homescreen/commit/b164558b9d8de74f41a011b0ded56e08d63b08da