toyota-connected/ivi-homescreen #144: The build error of ivi-homescreen_2.0

WAKazuyoshiAkiyama, 2024-11-08:
meta-flutter revision: `c4005faba9b0aae3f83cd4ef27e4bd936c4cd4cc` (`kirkstone`)
ivi-homescreen revision: 
```
HOMESCREEN_COMMIT ??= "25b67b5a32cce10daa3f20e20d00bf0acfc27fac"
PLUGINS_COMMIT ??= "875f40008e489c9e597694507622d5884eeed526"
```

PACKAGECONFIG:
```
PACKAGECONFIG = " \
    backend-wayland-egl \
    \
    egl-3d \
    egl-transparency \
    egl-multisample \
    \
    client-ivi-shell \
    client-xdg \
    \
    nav_render_view \
    \
    wdt \
    wdt_systemd \
    \
    dlt \
"
```

When I built ivi-homescreen_2.0 with the above configuration for `genericx86-64`, I got the following build error at do_compile.

```
ivi-homescreen/2.0-r0/git/third_party/waypp/src/window_manager/iv
i_window_manager.cc:17:10: fatal error: 'ivi_window_manager.h' file not found
#include "ivi_window_manager.h"
         ^~~~~~~~~~~~~~~~~~~~~~
1 error generated.
[...]
ivi-homescreen/2.0-r0/git/third_party/waypp/src/window_manager/registrar.cc:665:13: error: member reference type 'struct ivi_wm *' is a pointer; did you mean to use '->'?
  r->ivi_wm_.ivi_wm = static_cast<struct ivi_wm*>(wl_registry_bind(
  ~~~~~~~~~~^
            ->
[...]
ivi-homescreen/2.0-r0/git/third_party/waypp/src/window_manager/registrar.cc:665:13: error: incomplete definition of type 'ivi_wm'
  r->ivi_wm_.ivi_wm = static_cast<struct ivi_wm*>(wl_registry_bind(
  ~~~~~~~~~~^
[...]
ivi-homescreen/2.0-r0/git/third_party/waypp/src/window_manager/registrar.cc:667:63: error: member reference type 'struct ivi_wm *' is a pointer; did you mean to use '->'?
  LOG_DEBUG("{}: {}", interface, ivi_wm_get_version(r->ivi_wm_.ivi_wm.value()));
                                                    ~~~~~~~~~~^
                                                              ->
[...]
ivi-homescreen/2.0-r0/git/third_party/waypp/src/window_manager/registrar.cc:667:63: error: incomplete definition of type 'ivi_wm'
  LOG_DEBUG("{}: {}", interface, ivi_wm_get_version(r->ivi_wm_.ivi_wm.value()));
                                                    ~~~~~~~~~~^
```

WAKazuyoshiAkiyama, 2024-11-08:
I tied to add the following patch to resolve the error.

I could avoid the error of `'ivi_window_manager.h' file not found` with this patch, but, I could not fix other errors.

```
diff --git a/src/window_manager/ivi_window_manager.cc b/src/window_manager/ivi_window_manager.cc
index 92aa992..a089039 100644
--- a/src/window_manager/ivi_window_manager.cc
+++ b/src/window_manager/ivi_window_manager.cc
@@ -14,7 +14,7 @@
  * limitations under the License.
  */

-#include "ivi_window_manager.h"
+#include "waypp/window_manager/ivi_window_manager.h"

 #include "logging/logging.h"

--
2.44.1
```

jwinarske, 2024-11-09:
@WAKazuyoshiAkiyama `ivi-shell` is not currently supported on the v2.0 branch.