toyota-connected/ivi-homescreen #146: Couldn't able get the bundle folder

chandanagb16, 2024-12-04:
Trying to run the ivi-homescreen app with the plugins, followed the below steps to build the application

git clone --recurse-submodules -j8 https://github.com/toyota-connected/ivi-homescreen.git
git clone https://github.com/toyota-connected/ivi-homescreen-plugins.git
mkdir build && cd build
cmake ../ivi-homescreen -DCMAKE_STAGING_PREFIX=`pwd`/out/usr/local -DPLUGINS_DIR=`pwd`/../ivi-homescreen-plugins/plugins
make install -j

building is completed without any error and got an executable in the build/out/usr/local/bin/ folder with the name homescreen

but while running this homescreen app, it is expecting the bundle folder path, so can any one help on this how this bundle folder is created?

chandanagb16, 2024-12-06:
Please give some input on this

jwinarske, 2024-12-07:
@chandanagb16 The folder structure is documented in the README.md.
 
https://github.com/toyota-connected/ivi-homescreen#running-an-app

The bundle folder is a build artifact of either meta-flutter app recipe, workspace-automation, or by hand.
https://github.com/meta-flutter/workspace-automation

Also if you would like more interactive support please join the Discord channel.

chandanagb16, 2024-12-09:
@jwinarske 
I tried to install the workspace automation but getting fatal error for drm.h while installing flutter_workspace.py
I tried to install libdrm-dev package but still the  same issue observed
can you please help me if it is working for you

Is it not possible to create bundle folder with just having ivi-homescreen and ivi-homescreen-plugins 
because under cmake in  config_common.h file I could able to see the bundle path info

// Bundle folder paths
constexpr char kViewConfigToml[] = "config.toml";
constexpr char kBundleFlutterAssets[] = "data/flutter_assets";
constexpr char kBundleAot[] = "lib/libapp.so";
// Bundle folder override paths
constexpr char kBundleIcudtl[] = "data/icudtl.dat";
constexpr char kBundleEngine[] = "lib/libflutter_engine.so";

jwinarske, 2024-12-10:
Closing.  Discord discussion indicates a conflict with your amdgpu-emb installation.

closed 2024-12-10 by jwinarske