toyota-connected/ivi-homescreen #147: Don‘t support plugins for the aarch64 platform?

interestinghua, 2025-03-24:
I ran a Flutter application on the aarch64 platform and found that the application launched with flutter-auto cannot call the native method in the plugin through MethodChannel?
Is it currently not supported? Or did I compile it incorrectly?
Thanks！

jwinarske, 2025-03-25:
Hi.  Depends on what you're compiling.

1. It's not designed for dual pass builds (Flutter/Dart + Native).  It's optimized for single pass builds, as that's what is needed for embedded Linux support.  This will likely change when I finish the SDL3 integration; 3D games!
2. There are many native plugins that are supported today:
https://github.com/toyota-connected/ivi-homescreen-plugins/

If you find you need a plugin that is not present, let me know an I will consider adding it.

jwinarske, 2025-06-19:
@interestinghua can we close the issue?

closed 2025-07-29 by jwinarske